xjjak/LapCal #19: [FEATURE] PlatformIO environments for different handed gloves

xjjak, 2023-12-13:
For a second glove, that is supposed to be a right-hand one, there would need to be changes to the implementation of the sensors and calibration configuration.

PlatformIO environments and flash parameters could be utilized to specify the targeted hand so that configurations and the reversing of the order of sensors are changed accordingly.
For the calibration values should be stored in external files and only be inserted during compilation (like the Wi-Fi credentials) using said environments.

palisn, 2023-12-15:
If we want to use the same model, we will probably have to mirror/negate some values, at least the ones that involve lateral movement, as not doing so may imply incorrect predictions at worst and predicting some keys in the wrong order at best. For consistency I would imagine negating all the x- and y-accelerations as well as z- and possibly x-axis rotation might yield the best results.

xjjak, 2024-02-09:
I initially tried flipping the order of the sensors using two different for loops. This is way too unclean, so here is what to do next:
- [ ] Change `bool sensor_presence[]` to store `int`s and store the mux channel ids
- [ ] Iterate over sensor array
- [ ] `-1` -> sensor's missing
- [ ] `setup_sensors()` for use in `main.cpp`

xjjak, 2024-02-09:
@palisn You previously proposed to *negate*:
- *x, y, accel*
- *x rotation*(?)

Is this the first final decision or do you recommend any other changes? 
We'll be training a different model per hand anyway though, right? Would this still be necessary then?

palisn, 2024-02-12:
> Is this the first final decision or do you recommend any other changes? We'll be training a different model per hand anyway though, right? Would this still be necessary then?

I can't think of any other necessary changes right now, but even if we train a different model per hand, I would still keep the change, as we might be able to use data from the other hand respectively to improve the training process.

closed 2024-02-13 by xjjak