xjjak/LapCal #49: [IDEA] Restructure software directory

palisn, 2024-03-22:
**Describe the vision. What is to be achieved? What does the end-result look like? Is it a specific feature?**
The current structure is too thrown-together.  You can neither easily navigate it nor extend it properly.  The goal should be a clean structure that is easily extensible and allows for quick testing and development, as well as deployment.

**Share possible solutions or ideas and approaches.**
The top-level of the `Software` directory should provide four stages of software:
- Everything concerned with _data collection_ that is not hardware/firmware related. (`collection`)
- All _preprocessing_ of the collected data, as well as data format conversions. (`processing`)
- All _models_ (of every kind) and all related _custom modules_ that provide functionality for those models. (`machine-learning`)
- Everything related to _further processing_ of model output or other data, e.g. visualizations. (`analysis`)
Furthermore, we should embrace environment variables for the data storage location.

Additionally, we should replace the central `requirements.txt` with a more decentralized approach, that requires only exactly as much as needed for a specific tool (my idea is to leave the `requirements.txt` in place, but also provide `shell.nix` files for every tool).  Along with that, every tool and most directories should get their own READMEs that explain their usage and other important information regarding them.

**Additional context**
The current structure strongly differentiates between click and state-detection.  They also both rely on different data collection method, with no clean way to interoperate them.

xjjak, 2024-03-22:
Should a folder dedicated to machine learning *on-the-edge* (`edge-ml`?) be added to the 2nd level or put inside the `machine-learning` folder?

palisn, 2024-03-23:
> Should a folder dedicated to machine learning _on-the-edge_ (`edge-ml`?) be added to the 2nd level or put inside the `machine-learning` folder?

Wouldn't that be firmware?

xjjak, 2025-01-14:
Regarding the proposed migration away from `requirements.txt`: Would [uv](https://github.com/astral-sh/uv) be an alternative/additional option for you, for managing python scripts etc.? @palisn 
As far as I understand it is also still compatible with pip, but makes managing `venvs` and dependencies much easier.

PR #68 (2025-05-23): Restructure software directory
This endeavor has taken up too much time and energy to actually warrant the end result, thus, we instead opted to remove all software parts that are not currently used and initiate the new structure just for new and still relevant stuff. If in the future any of the old stuff is needed again, it should be incorporated into the new structure then.

The new structure also differs quite a bit from the initial idea. The new idea is a lot looser requiring only the following:
- The software should broadly be **centered around a small number of main org mode files** (cur [... cut]
closed 2025-05-23 by xjjak