xjjak/LapCal #60: [ERROR] Live-plotting: "inclusive-range: unbound identifier"

xjjak, 2024-10-12:
**Describe the problem**
When trying to run the live-plotting racket program an error is returned.
This is the case in both the `live-plotting-development` as well `main` branch.

May have something to do with my Racket installation.

```
src/gui/plot-canvas.rkt:118:40: closed-range: unbound identifier
  in: closed-range
  location...:
   src/gui/plot-canvas.rkt:118:40
  context...:
   /usr/share/racket/collects/racket/private/stxparam.rkt:61:2: let-local-keys
   [repeats 3 more times]
   /usr/share/racket/collects/racket/private/class-undef.rkt:35:0: detect-field-unsafe-undefined
```

palisn, 2024-10-12:
I'll look into it, but the error message you provided does not match the title. I would guess you probably already tried to fix it and forgot to revert it?

palisn, 2024-10-12:
What racket version (

xjjak, 2024-10-12:
> I'll look into it, but the error message you provided does not match the title. I would guess you probably already tried to fix it and forgot to revert it?

Yeah, whoops. Only tried that one substitution after looking into the documentation on ranges.

xjjak, 2024-10-12:
> @xjjak What racket version (`racket --version`) are you using?

Updated issue description. Using version 7.9

palisn, 2024-10-12:
That seems like it's the issue. Documentation says
> Added in version 8.0.0.13 of package base.

So I guess updating should probably fix the issue. I'm working on a proper project setup right now anyway, so that will hopefully fix such issues in the future.

closed 2024-10-15 by xjjak